Xinyuan-LilyGO/T-Impulse #1: error: 'struct ADC_InitTypeDef' has no member named 'OversamplingMode'

maikebing, 2021-11-15:
\LilyGo-T-Impulse\Example\ArduinoDemo\T-Wristband-S76G\T-Wristband-S76G.ino: In function 'void MX_ADC_Init()':
T-Wristband-S76G:792:13: error: 'struct ADC_InitTypeDef' has no member named 'OversamplingMode'
  792 |   hadc.Init.OversamplingMode = DISABLE;
      |             ^~~~~~~~~~~~~~~~
T-Wristband-S76G:795:13: error: 'struct ADC_InitTypeDef' has no member named 'SamplingTime'; did you mean 'SamplingTimeCommon'?
  795 |   hadc.Init.SamplingTime = ADC_SAMPLETIME_39CYCLES_5;
      |             ^~~~~~~~~~~~
      |             SamplingTimeCommon
T-Wristband-S76G:795:28: error: 'ADC_SAMPLETIME_39CYCLES_5' was not declared in this scope; did you mean 'ADC_SAMPLETIME_239CYCLES_5'?
  795 |   hadc.Init.SamplingTime = ADC_SAMPLETIME_39CYCLES_5;
      |                            ^~~~~~~~~~~~~~~~~~~~~~~~~
      |                            ADC_SAMPLETIME_239CYCLES_5
T-Wristband-S76G:806:13: error: 'struct ADC_InitTypeDef' has no member named 'LowPowerFrequencyMode'
  806 |   hadc.Init.LowPowerFrequencyMode = DISABLE;
      |             ^~~~~~~~~~~~~~~~~~~~~
T-Wristband-S76G:818:38: error: 'ADC_SINGLE_ENDED' was not declared in this scope
  818 |   HAL_ADCEx_Calibration_Start(&hadc, ADC_SINGLE_ENDED);
      |                                      ^~~~~~~~~~~~~~~~
 LilyGo-T-Impulse\Example\ArduinoDemo\T-Wristband-S76G\T-Wristband-S76G.ino: In function 'void BoardInit()':
T-Wristband-S76G:916:15: error: 'IICSCL' was not declared in this scope
  916 |   Wire.setSCL(IICSCL);
      |               ^~~~~~
T-Wristband-S76G:917:15: error: 'IICSDA' was not declared in this scope
  917 |   Wire.setSDA(IICSDA);
      |               ^~~~~~
 
exit status 1
'struct ADC_InitTypeDef' has no member named 'OversamplingMode

LilyGO, 2021-11-16:
Please make sure that the STM32 environment is installed in your arduino, ADC_InitTypeDef and other content are all in the STM32 SDK

closed 2021-11-22 by LilyGO

AbdelsalamHaa, 2023-06-14:
Hi @LilyGO 
I have the same issue. I'm quite sure I installed everything correctly as the guide line. 
When I use the example GPStest everything works well. But when using S76G-Factory I get the same issue.

here is the error message I got 

      C:\Users\Abdelsalam\Downloads\Lora_Tracker_Watch\Lora_Tracker_Watch.ino: In function 'void MX_ADC_Init()':
      C:\Users\Abdelsalam\Downloads\Lora_Tracker_Watch\Lora_Tracker_Watch.ino:800:13: error: 'struct ADC_InitTypeDef' has no member named 'OversamplingMode'
        800 |   hadc.Init.OversamplingMode = DISABLE;
            |             ^~~~~~~~~~~~~~~~
      C:\Users\Abdelsalam\Downloads\Lora_Tracker_Watch\Lora_Tracker_Watch.ino:803:13: error: 'struct ADC_InitTypeDef' has no member named 'SamplingTime'; did you mean 'SamplingTimeCommon'?
        803 |   hadc.Init.SamplingTime = ADC_SAMPLETIME_39CYCLES_5;
            |             ^~~~~~~~~~~~
            |             SamplingTimeCommon
      C:\Users\Abdelsalam\Downloads\Lora_Tracker_Watch\Lora_Tracker_Watch.ino:803:28: error: 'ADC_SAMPLETIME_39CYCLES_5' was not declared in this scope; did you mean 'ADC_SAMPLETIME_239CYCLES_5'?
        803 |   hadc.Init.SamplingTime = ADC_SAMPLETIME_39CYCLES_5;
            |                            ^~~~~~~~~~~~~~~~~~~~~~~~~
            |                            ADC_SAMPLETIME_239CYCLES_5
      C:\Users\Abdelsalam\Downloads\Lora_Tracker_Watch\Lora_Tracker_Watch.ino:814:13: error: 'struct ADC_InitTypeDef' has no member named 'LowPowerFrequencyMode'
        814 |   hadc.Init.LowPowerFrequencyMode = DISABLE;
            |             ^~~~~~~~~~~~~~~~~~~~~
      C:\Users\Abdelsalam\Downloads\Lora_Tracker_Watch\Lora_Tracker_Watch.ino:824:38: error: 'ADC_SINGLE_ENDED' was not declared in this scope
        824 |   HAL_ADCEx_Calibration_Start(&hadc, ADC_SINGLE_ENDED);
            |                                      ^~~~~~~~~~~~~~~~
      
      exit status 1
      
      Compilation error: 'struct ADC_InitTypeDef' has no member named 'OversamplingMode'