Xinyuan-LilyGO/T-Impulse #2: T-Impulse Demo doesn't work

gadLinux, 2021-11-15:
I can build and upload the demo but it doesn't seem to do anything.

Nothing shown in the screen and nothing in TTN...

Should I do something special?

LilyGO, 2021-11-16:
There are a few suggestions:
1. T-Impule Demo is built with the VS CODE plug-in platfromIO. Make sure that this plug-in is installed on your computer and updated to the latest version.
2. If you want to use the function of LoRaWAN, please make sure you have a gateway device connected to TTN.

gadLinux, 2021-11-16:
Hi, 
The situation is that I can build, deploy and burn the artefact (Firmware) but the t-impulse watch remains black. The usb device changes from DFU to Bus 003 Device 059: ID 0483:5740 STMicroelectronics Virtual COM Port

But no signal of activity in it. I think it's not correctly burned or software bug. I will try to give more details. My doubt is. Should it work out-of-the box by just using PlatformIO and burning?

I updated loramac file... Gateway is on and running...

Could you provide any detail, please?

gadLinux, 2021-11-16:
Ok. Seen that Over the terminal port I can see the logs, and It's trying to join. So nothing is seen on screen and that's normal... Resolved. Can I make a PR explaining?

gadLinux, 2021-11-16:
Today it worked... I don't know what changed... But it shows a little text, and a couple of icons.
Selected env:nucleo_l073rz and burned...

gadLinux, 2021-11-16:
Not every time I burn the image it works...

gadLinux, 2021-11-16:
Ok, there's some issues with the GPS if I get out the GPS initialization it works. I removed the GPS loop too..

LilyGO, 2021-11-18:
Hello, after uploading using PLATFROMIO, it should be recommended to press the restart button on the back of the shell again. I also tried the black screen when uploading, but it was normal after resetting. This may be a bug in the uploading program.

LilyGO, 2021-11-18:
Can you talk about GPS in detail?

gadLinux, 2021-11-18:
Yes, I commented out some lines in the main.cpp file to make it work.

void BoardInit(void)
{
    Serial.begin(115200);
    delay(1000);
    Serial.println("Basic init ok");
    pinMode(PwrSwitch1_8V, OUTPUT);
    digitalWrite(PwrSwitch1_8V, HIGH);
    pinMode(PwrSwitchGPS, OUTPUT);
    digitalWrite(PwrSwitchGPS, HIGH);
    Wire.setSCL(IICSCL);
    Wire.setSDA(IICSDA);
    Wire.begin();

//    GPS_Init();
    OledInit();
    IMUInit();

 
    Bat_Init();
    pinMode(BatteryVol, INPUT_ANALOG);

    LoraWanInit();

    pinMode(TTP223_VDD_PIN, OUTPUT);
    pinMode(TouchPad, INPUT);
    digitalWrite(TTP223_VDD_PIN, HIGH); 
}

void loop()
{
     loopLMIC();
    Bat_loop();
    //GPS_loop(); 
    Serial.println("Loop done");
    delay(1000);
}

Now it boots 100% of time. I had not time to investigate why it doesn't boot with the GPS stuff in

gadLinux, 2021-11-20:
It seems gpsPort.available() never returns data... So GPS maybe wrong or unavailable...

gadLinux, 2021-11-20:
I have reinstalled the Firmware that's in the binary form

And I can see this in the log...
GPS Send cmd Fail                                                                                                                                                 
GPS Send cmd Fail                                                                                                                                                 
GPS Send cmd Fail                                                                                                                                                 
GPS Send cmd Fail                                                                                                                                              
GPS Send cmd Fail                                                                                                                                                 
GPS done        

https://github.com/Xinyuan-LilyGO/LilyGo-T-Impulse/blob/master/Example/ArduinoDemo/TestFirmware/firmware.bin

This is because it sends the 5 messages, fail at each one and then it continues...
After that no GPS fix... I left for overnight so I'm starting to think there's a problem on the GPS hardware...

gadLinux, 2021-11-20:
WARNING: No GPS data. Check wiring.

When on GPS screen

gadLinux, 2021-11-20:
Can you point me where is it the GPS module, because I cannot find it.
[image]
.
[image]
.

gadLinux, 2021-11-20:
Addtitionally SoftRF has a great built-in check:

Built-in components:                                                                                                                                              
RADIO   : PASS                                                                                                                                                    
CLK SRC : TCXO                                                                                                                                                    
GNSS    : FAIL                                                                                                                                                    
DISPLAY : PASS                                                                                                                                                    
IMU     : PASS   

So I think GPS is dead in this device... And that's causing all troubles... Maybe I can try to resolder, but I contacted Lillygo official store just in case...

LilyGO, 2021-11-22:
1. GPS module has been integrated in that big chip.
2. Can you assist in writing such a program on the device, so that I can check if the GPS is working?  https://github.com/Xinyuan-LilyGO/LilyGo-T-Impulse/tree/master/Example/GPStest

gadLinux, 2021-11-22:
I was looking at the specs, and I realized that so it doesn't seems like it can be broken. It means that initialization was not done right. So let's look a little bit at it.

gadLinux, 2021-11-22:
I can help with it, did a fork and PR. Maybe you can create a branch and I will do push to that branch so we don't mess with master
https://github.com/Xinyuan-LilyGO/LilyGo-T-Impulse/pull/5

gadLinux, 2021-11-22:
I'm thinking remove battery to see if leaving it without power does a reset of internal capacitors...

LilyGO, 2021-11-23:
After uploading the test firmware, if there is no information about GPS returned, I guess the crystal oscillator is broken, you can contact customer service

LilyGO, 2021-11-23:
The position of the crystal oscillator is behind the OLED screen

gadLinux, 2021-11-23:
Any ideas on how to solve this trouble? Do you know if a chip reset will do the trick? Is there any way to recover the functionality?

closed 2021-11-24 by gadLinux

Pattepat, 2023-02-17:
@gadLinux I know this is a very old post, but i just received my t-impulse from lilygo and have the same issues with GPS.  Lora is working fine but the GPS is failing also. (No wiring message) and the SoftRF-init also shows a fail on GPS.  Tried many different examples, but GPS seems dead.

`Built-in components:
RADIO   : PASS
CLK SRC : TCXO
GNSS    : FAIL`

So was wondering if you ever got this (GPS) working or was your unit broken (like i think mine is)

Thanks

gadLinux, 2023-03-09:
Hi @Pattepat I'm finally getting serious about the project. I'm moving to rust so it will take me a lot to reach to that point. But I will tell you... I received the crystal but didn't installed yet. But I have another band to test. Will keep you posted.

Litosaragon, 2023-08-18:
Same problem here, any update?

philippehenin, 2023-09-22:
https://photos.app.goo.gl/F8nCZH8fxDc4Kdjo9

Please excuse the quality of the video, but when the watch stuck or frozen, I go outside when it can catch a satellite everything went back to normal...

Did you try?

Have a pleasant day,
P.