Xinyuan-LilyGO/T-Impulse #3: Cannot get the device to join TTN

gadLinux, 2021-11-16:
I'm receiving a message like this one after joining is attempted

4956739: Unknown event

It seems the packet is not received ok. Or just it had a problem. 
LMIC seems to receive the information but it's not identified as a correct packet

I see in the gateway:
JSON up: {"rxpk":[{"tmst":192234123,"chan":1,"rfch":1,"freq":868.300000,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","lsnr":9.8,"rssi":-17,"size":23,"data":"AAAXXAAAAAAAA5oQE017Vs3ALVVKPuf4="}]}

And in the TTN device Accept Join request.

I'm using the semtech packet forwarder that goes with my gateway (https://github.com/Lora-net/packet_forwarder). The one indicated in the setup page doesn't work for me since it seems gpio are not the same.

LilyGO, 2021-11-18:
Do you mean that the gateway has received the network access information, but it does not show that the connection is successful on the TTN?

gadLinux, 2021-11-18:
The join request arrives to the gateway and from the gateway to the TTN network. That works. What it doesn't work is the response from the network to device. Message arrvies to the device but it's not understood.

More in detail:

Following this guide:

    https://github.com/will127534/LoRa-concentrator/wiki/Startup-Guide
    The Gateway is connected to TTN v3 and it shows it's live!
    https://ttnmapper.org/heatmap/private/?gateway=fe000100ff000001&network=NS_TTS_V3://ttn@000013
    So that's a good thing
    [imagen](https://user-images.githubusercontent.com/233204/142183002-8e82da16-7cd6-4090-bd3c-68997d08c690.png)

I can see Join request on TTN and in the gateway

JSON down: {"txpk":{"imme":false,"tmst":463689596,"freq":868.1,"rfch":0,"powe":14,"modu":"LORA","datr":"SF8BW125","codr":"4/5","ipol":true,"size":33,"ncrc":true,"data":"IOwUXX8rfaim5TUqK6X0jPGtg6gF4Xz0Q/16C4DLDq/P"}}

But it fails on device... because the kind of message is not expected.
I attach some gateway logs:

Some logs too:

```
##### 2021-11-17 10:30:57 GMT #####
### [UPSTREAM] ###
# RF packets received by concentrator: 0
# CRC_OK: 0.00%, CRC_FAIL: 0.00%, NO_CRC: 0.00%
# RF packets forwarded: 0 (0 bytes)
# PUSH_DATA datagrams sent: 1 (113 bytes)
# PUSH_DATA acknowledged: 100.00%
### [DOWNSTREAM] ###
# PULL_DATA sent: 6 (100.00% acknowledged)
# PULL_RESP(onse) datagrams received: 0 (0 bytes)
# RF packets sent to concentrator: 0 (0 bytes)
# TX errors: 0
# TX rejected (collision packet): 0.00% (req:1, rej:0)
# TX rejected (collision beacon): 0.00% (req:1, rej:0)
# TX rejected (too late): 0.00% (req:1, rej:0)
# TX rejected (too early): 100.00% (req:1, rej:1)
# BEACON queued: 0
# BEACON sent so far: 0
# BEACON rejected: 0
### [JIT] ###
# SX1301 time (PPS): 92277009
src/jitqueue.c:448:jit_print_queue(): INFO: [jit] queue is empty
### [GPS] ###
# Invalid time reference (age: 1637145057 sec)
# no valid GPS coordinates available yet
##### END #####
##### 2021-11-17 10:34:57 GMT #####
### [UPSTREAM] ###
# RF packets received by concentrator: 0
# CRC_OK: 0.00%, CRC_FAIL: 0.00%, NO_CRC: 0.00%
# RF packets forwarded: 0 (0 bytes)
# PUSH_DATA datagrams sent: 1 (156 bytes)
# PUSH_DATA acknowledged: 100.00%
### [DOWNSTREAM] ###
# PULL_DATA sent: 6 (100.00% acknowledged)
# PULL_RESP(onse) datagrams received: 0 (0 bytes)
# RF packets sent to concentrator: 0 (0 bytes)
# TX errors: 0
# TX rejected (collision packet): 0.00% (req:3, rej:0)
# TX rejected (collision beacon): 0.00% (req:3, rej:0)
# TX rejected (too late): 0.00% (req:3, rej:0)
# TX rejected (too early): 33.33% (req:3, rej:1)
# BEACON queued: 0
# BEACON sent so far: 0
# BEACON rejected: 0
### [JIT] ###
# SX1301 time (PPS): 332276061
src/jitqueue.c:448:jit_print_queue(): INFO: [jit] queue is empty
### [GPS] ###
# Invalid time reference (age: 77 sec)
# GPS coordinates: latitude 40.57798, longitude -3.92264, altitude 835 m
##### END #####
##### 2021-11-17 10:34:57 GMT #####
### [UPSTREAM] ###
# RF packets received by concentrator: 0
# CRC_OK: 0.00%, CRC_FAIL: 0.00%, NO_CRC: 0.00%
# RF packets forwarded: 0 (0 bytes)
# PUSH_DATA datagrams sent: 1 (156 bytes)
# PUSH_DATA acknowledged: 100.00%
### [DOWNSTREAM] ###
# PULL_DATA sent: 6 (100.00% acknowledged)
# PULL_RESP(onse) datagrams received: 0 (0 bytes)
# RF packets sent to concentrator: 0 (0 bytes)
# TX errors: 0
# TX rejected (collision packet): 0.00% (req:3, rej:0)
# TX rejected (collision beacon): 0.00% (req:3, rej:0)
# TX rejected (too late): 0.00% (req:3, rej:0)
# TX rejected (too early): 33.33% (req:3, rej:1)
# BEACON queued: 0
# BEACON sent so far: 0
# BEACON rejected: 0
### [JIT] ###
# SX1301 time (PPS): 332276061
src/jitqueue.c:448:jit_print_queue(): INFO: [jit] queue is empty
### [GPS] ###
# Invalid time reference (age: 77 sec)
# GPS coordinates: latitude 40.57798, longitude -3.92264, altitude 835 m
##### END #####
```

But it can something wrong in the device because cannot handle the message:

Device fails to parse the join response with:

4956739: Unknown event

gadLinux, 2021-11-20:
I added more logs to the device And this is what it shows:

757063: setupRx1 txrxFlags 00 --> 01                                                                                                                              
start single rx: now-rxtime: 60517                                                                                            
757500: RXMODE_SINGLE, freq=868500000, SF=7, BW=125, CR=4/5, IH=0                           
Loop done                  
rxtimeout: entry: 760414 rxtime: 696868 entry-rxtime: 63546 now-entry: 59786 rxtime-txend: 311126                                                                 
Loop done                                                        
883063: setupRx2 txrxFlags 0x1 --> 02                                                                                                                             
start single rx: now-rxtime: 124270                                                                                          
883510: RXMODE_SINGLE, freq=869525000, SF=12, BW=125, CR=4/5, IH=0                        
Loop done               
rxtimeout: entry: 899802 rxtime: 759118 entry-rxtime: 140684 now-entry: 46461 rxtime-txend: 373376                                                                
946635: processRx2Jacc txrxFlags 0x2 --> 00                     
946787: Unknown event
Loop done                                                                                                                                                         
1009251: engineUpdate, opmode=0x4                                                                                                                                 

Seems like it's not receiving answer after the two frame wait period...

gadLinux, 2021-11-20:
Found the problem!
Code works perfectly. The problem was derived of the GPS trouble in the other issue and my fault. 
I added a delay in loop to be able to see logs more easily and this delayed all processing avoiding the device to join because no response was detected... I'm sorry about that.

closed 2021-11-20 by gadLinux