Xinyuan-LilyGO/T-Impulse #7: #define CFG_au915 1

tcpipchip, 2022-01-14:
Hi
Enabling CFG_au915 on LMIC_PROJECT_CONFIG.H

`//#define CFG_eu868 1
//#define CFG_us915 1
#define CFG_au915 1
//#define CFG_as923 1
// #define LMIC_COUNTRY_CODE LMIC_COUNTRY_CODE_JP      /* for as923-JP; also define CFG_as923 */
//#define CFG_kr920 1
//#define CFG_in866 1`

And making making the Setup of loramac.cpp

I am getting error

[image]

Can you verify ?

if i back to #define CFG_eu868 1

Compiles!

[image]

tcpipchip, 2022-01-14:
//LMIC_setupChannel(0, 868100000, DR_RANGE_MAP(DR_SF12, DR_SF7), BAND_CENTI);  // g-band
    //LMIC_setupChannel(1, 868300000, DR_RANGE_MAP(DR_SF12, DR_SF7B), BAND_CENTI); // g-band
    //LMIC_setupChannel(2, 868500000, DR_RANGE_MAP(DR_SF12, DR_SF7), BAND_CENTI);  // g-band
    //LMIC_setupChannel(3, 867100000, DR_RANGE_MAP(DR_SF12, DR_SF7), BAND_CENTI);  // g-band
    //LMIC_setupChannel(4, 867300000, DR_RANGE_MAP(DR_SF12, DR_SF7), BAND_CENTI);  // g-band
    //LMIC_setupChannel(5, 867500000, DR_RANGE_MAP(DR_SF12, DR_SF7), BAND_CENTI);  // g-band
    //LMIC_setupChannel(6, 867700000, DR_RANGE_MAP(DR_SF12, DR_SF7), BAND_CENTI);  // g-band
    //LMIC_setupChannel(7, 867900000, DR_RANGE_MAP(DR_SF12, DR_SF7), BAND_CENTI);  // g-band
    //LMIC_setupChannel(8, 868800000, DR_RANGE_MAP(DR_FSK, DR_FSK), BAND_MILLI);   // g2-band
    LMIC_selectSubBand(2); //TTN

i have to do this!

tcpipchip, 2022-01-14:
now publishing on TTN

tcpipchip, 2022-01-14:
Need to study no CAYENNE

tcpipchip, 2022-01-14:
Broken link
https://github.com/Xinyuan-LilyGO/LilyGo-T-Impulse/blob/5fc3b7bf6f32c115bf8c6d9ce80eae6b6e94ee50/Example/T-Impulse%20Demo/README.md

tcpipchip, 2022-02-12:
Solved! Cayenne working nice now!

LilyGO, 2022-02-23:
Thank you for your spirit of discovery and exploration

closed 2022-02-23 by LilyGO