Xinyuan-LilyGO/T-Impulse #12: s76g lora example not working

Wailshire, 2022-11-25:
hello, i just bought one s76g and uploaded the code, all is working perfectly except the lora, i cant tx or rx,  we need to change something in the config to make it work?

mmMicky, 2022-11-26:
Lora needs to be configured with the same frequency as the spreading factor to be able to transmit point-to-point. You can try uploading factory routines. Select sending and receiving respectively to test.

realhik, 2023-03-01:
You have to alter the definitions in `loramac.cpp` as explained in [this description](https://github.com/Xinyuan-LilyGO/T-Impulse/tree/master/Example/T-Impulse-S76G-LoRaWAN). Make sure that you choose  `lsb` / `msg` before copy & paste the keys.