Xinyuan-LilyGO/T-Impulse #15: GPS serial communication fails

realhik, 2023-02-28:
I have tested the T-Impulse and got GPS and LoRaWAN working for TTN.

However, since a few days I could no longer get a GPS fix. Today, I have found out that the serial communication to the GNSS stopped working. Or to be more precise: all `GPS_WaitAck("...");`commands are no longer answered. All other peripherals work as expected.  I used the original code with an `Serial.println(ack);` statement after `ack = gpsPort.readStringUntil('\n')` in `GPS_Waitack();`.

Any ideas how this can be fixed?

Thanks
Thorsten

realhik, 2023-02-28:
Could the problem possibly be due to missing 1.8V, so that the GNSS has no supply voltage? How can this be checked?

Litosaragon, 2023-08-18:
Same problem here, any update?

pmoncks, 2023-08-18:
I'm also facing the same issue. Any solution?

Litosaragon, 2023-08-18:
If someone have any update about this issue, please write here in order to have all the solution ASAP