Xinyuan-LilyGO/T-Impulse #18: T Impulse Not Detected In Windows

ajgrinds, 2023-07-17:
Hello Everyone,

I am having this problem where my T impulse is only detected using 1 specific USB-C cable. It doesn't even work with the cable provided in the box. Does anyone know what kind of cable I need? Im assuming its something with high power but I cannot find anything online.

dforco, 2023-07-23:
in my case it worked only once (i could upload G76 example with Arduino IDE). Then the STM32 boatloader seams to broke and COM4 interface did not appeared anymore. Is there any way to access UART communication to try to burn an USB Boatloader again?

ajgrinds, 2023-07-23:
Mine works multiple times but only with one specific cable. But yes you should be able to prove the UART to flash that way but if the stm32 boot loader is dead you’re SOL.

dforco, 2023-07-24:
Thank you Andrew! and how do I access to UART tx/rx pins? I couldnt find any hardware documentation in the web

jp-bennett, 2023-08-16:
@ajgrinds Hey, stumbled across this while working on my newly-received T-impulse. The solution here is that the clear plastic cover has to be removed, to get the USB-c cable to fully seat. An extended tip USB-c cable would probably work too. I'm guessing that quite a few people have had this particular problem with this device.

ajgrinds, 2023-08-16:
It still doesn't work for me. My cables have D+ and power + shorted
together to get it to work. The board is designed incorrectly, there isn't
supposed to be a pullup outside of the UART chip as it is already included
in the chip (therefore having 2 pullups)

Ferdiethecat, 2024-07-04:
I am also unable to detect the device on USB. I noted that not all USB cables would charge the device so assumed the connector spring force/PCB alignment/ engagement length  could have been the initial issues.  I don't see the 'pull-up' on circuit diagram but have not examined the unit PCB.  I would think (or at least hope) that the USB would have been used to load the factory demo programme so. Has anyone found a fix or that a second device works fine?